rusefi/rusefi #2953: CEL to indicate WBO error

BeerMoneyMotorsports, 2021-07-12:
We need to implement The CEL to relay WBO errors. We have no gauge to assist in this, so CEL implementation is critical. It would be great to tie it to the blue error led so the error will be shown in CEL blinking and can be identified without opening the case. At minimum, flashing CEL for any wbo error is required.

BeerMoneyMotorsports, 2021-07-13:
This should include a toggle in TS to shut it off for those NOT running internal WBO.

mck1117, 2021-07-13:
We can do it for some external WBO as well - both AEM xseries and us indicate invalid (not some random AFR reading) when the sensor is warming up, overheated, or missing.  As for our WBO controller, internal is exactly the same as external.

BeerMoneyMotorsports, 2021-08-13:
@mck1117 poke poke.

BeerMoneyMotorsports, 2021-12-14:
is this on the docket at all?  @rusefillc @mck1117