rusefi/rusefi #2980: NB2 trigger shape does not seem to match JimStim

rusefillc, 2021-07-17:
JimStim does not approve of rusEFI trigger shape definition? note errors in coil scheduling

depending on RPM we either have one coil event not even or just totally not present

[image]

[image]

rusefillc, 2021-07-17:
[image]

rusefillc, 2021-07-17:
[image]

rusefillc, 2021-07-17:
and now self-stim also show uneven sparking, so the problem is NOT trigger definition?!

[image]

rusefillc, 2021-07-17:
new debug gauge in the "instance RPM" family confirms problem

this log is with external jimstim

[image]

mck1117, 2021-07-18:
trigger is correct, prepareEventAngles is broken

PR #2982 (2021-07-18): Fix prepareEventAngles for symmetrical crank triggers
fixes #2980
probably also fixes #2958
closed 2021-07-18 by rusefillc