rusefi/rusefi #3003: Nissan 360 CAS

rusefillc, 2021-07-21:
At the moment rusEFI does not support Nissan 360 CAS for multiple reasons

rusefillc, 2021-07-21:
This is a low priority due to above average level of effort for this specific wheel

We already support just the 3+1 second channel and we already have an engine running like that.

rusefillc, 2021-07-21:
```
// as of April 2020, trigger which requires most array length is REMIX_66_2_2_2
// we can probably reduce RAM usage if we have more custom logic of triggers with large number of tooth while
// pretty easy logic. like we do not need to REALLY have an array to remember the shape of evenly spaces 360 or 60/2 trigger :)
#define PWM_PHASE_MAX_COUNT 280
```

rusefillc, 2021-07-21:
OMG they come in DIFFERENT shapes?!

[image]

rusefillc, 2021-07-21:
VH41DE (Japaneese Y32 Variant)

mck1117, 2021-07-21:
> OMG they come in DIFFERENT shapes?!

yes - the inner ring has one hole per cylinder, so there are at least i4/v6/v8 variants

kjngineering, 2021-07-21:
so the interleaved slots I think are 2deg, 3deg, 4deg and 5deg for this particular wheel, I will get my CAS and do a test for you guys!

mck1117, 2021-07-21:
Most sure-fire method is to spin the wheel at speed (drill? lathe?), and grab the output of the sensor with a logic analyzer, if that's an option.

rusefillc, 2021-07-21:
https://speeduino.com/wiki/index.php/Nissan_360

[Nissan 360 - Speeduino.pdf](https://github.com/rusefi/rusefi/files/6852166/Nissan.360.-.Speeduino.pdf)