rusefi/rusefi #3005: ETC blip for downshift

rusefillc, 2021-07-21:
TBD

rusefillc, 2021-07-22:
load cell and amplifier
0-5v analog

2.5v neutral
3v means up-shift
2v means down-shift

rusefillc, 2021-07-22:
histeresis not needed, just the edge crossing is good enought

and then we have a duration for blip and duration of retard

mmatousek89, 2021-07-22:
blip - downshift

analog input below 2V to trigger blip event, blip event is 20% etc position for 150ms
arming time prevent another blip in less than 300ms from last blip

condition:
pedal position sensor below 1% (idealy engine torque is below 0Nm)
engine rpm below 7000

hints:
antilag can produce positive torque, in this case undercut on dogs inside gearbox does not allow to pull gear out and throtlle blip (which add more positive torque) does not help at all.
if we can sense engine torque and see that pilot is trying to execute downshift but torque is already positive we should perform  spark cut/retard to reduce torque below 0 to release gearbox dogs, together with throttle blip which add positive torque to help rise engine speed to match with driveshaft speed for new lower gear to engange

rusefillc, 2021-08-08:
@mck1117 opinions - brute force implementation for now I guess?

CONFIG(gearboxPressureSensorAnalogInput)
CONFIG(blipAmountPercent)
CONFIG(blipDurationMs)
CONFIG(armingTime)

rusefillc, 2024-02-17:
Lua is recommended here before we put into main firmware

rusefillc, 2025-04-04:
@ElDominio can we reuse or do you say we start from scratch?

ElDominio, 2025-04-04:
I would say start from scratch, since I have a separate way of understanding how this should work.

rusefillc, 2025-04-04:
Moving to https://github.com/rusefi/rusefi/issues/7681