rusefi/rusefi #3006: gear spark retard+cut for upshift

rusefillc, 2021-07-21:
TBD

mmatousek89, 2021-07-22:
gear spark cut - upshift

strategy for dogbox gearbox only, preferably sequential, aplicable for H pattern too

analog input from gear lever force transducer will tell ecu that there is demand from driver to upshift
this can happen only when driver is on gas, ie pedal position sensor over 30%, in this time there is positive flow of torque from engine to driveshaft and this torque due to dogbox undercut holds gear in , and driver cant pull gearstick any further to perform upshift, so analog input is rising to threshold for example 3Volts , where ecu perform ignition timing retard or cut some sparks for 80-120ms, result of this is rapid torque reduction which move gears in gearbox from locked position, gear lever can move and engange higher gear.

rusefillc, 2021-08-15:
Depends on #3159

mck1117, 2022-02-16:
adequate lua hooks exist for this