rusefi/rusefi #3008: Come up with path for compatibility with AIM devices

stefanst, 2021-07-21:
Currently rusEFI FW and AIM displays do not communicate.
The solution is presumably to talk to AIM and get them to incorporate a configuration for rusEFI devices.

rusefillc, 2021-07-21:
@stefanst please explain like I am five

is there a URL for AIM? what specific devices are we talking about?

Do you own an AIM display unit?

stefanst, 2021-07-21:
Unfortunately I am only aware of the broad outlines:
AIM devices are very popular with the more serious racers. They make displays, cameras, shift lights, data acquisition modules, GPS modules etc. that all interconnect. 
Using their setup software "Race Studio 3" you can select which ECU you have and then the proper CAN settings get selected automatically. 
@mck1117 has suggested that AIM compatibility would be as simple as talking to them and incorporating a definition for our rusEFI datastream in their setup options. 

I do not own an AIM device, since I am not a serious racer and the AIM devices are quite pricey. But I at least have access to one via Michael Hartigan and his race Miata (NA8). He is currently using MSPNP and it died this last weekend in the rain. If we had a setup, I could probably seize the opportunity and talk him into going rusEFI. 
So, at this point we at least have limited access to a car with AIM display and cameras for testing.

This may be an opportunity to appeal to a less cheap class of customer than our standard DIY fiddle-with-my-Miata guys.

mck1117, 2021-07-21:
We could also tx the AIM format as a dash option.  They're much less likely to support our protocol than we are to send theirs.

rusefillc, 2021-07-21:
My concern about our protocol is that I have too little confidence in us having it compatible for 5 years

Implementing someone's else protocol looks like a way more reliable option for 5 years

stefanst, 2021-07-21:
I just installed the Race Studio 3 software. The list of available protocols is impressive. I mean they even have "MV Agusta" bikes.
This is what you can select for MS:
[image]

rusefillc, 2021-11-27:
lol #3599

mck1117, 2022-02-16:
Hey @opelpanfan, you made progress here right?

mck1117, 2022-02-16:
deduplicating issues https://www.aimtechnologies.com/support/racingecu/AiM_CAN_101_eng.pdf

opelpanfan, 2022-02-16:
Yes fixed some dividers/multipliers and now works on AIM MXS Strada. PR #3924

closed 2026-10-21 by rusefillc