rusefi/rusefi #3010: GHA reliability story triggers/config

rusefillc, 2021-07-21:
what's going on here? triggers? config? I am confused

https://github.com/rusefi/rusefi/runs/3119814717?check_suite_focus=true

[image]

mck1117, 2021-07-21:
I think it might be:

- commit 1 triggers build A, build B
- build A commits file X
- build B tries to commit file, but master is already ahead, so it fails

rusefillc, 2021-07-21:
we already attempted something #2446 

```
        git config --local user.email "action@github.com"
        git config --local user.name "GitHub build-unit-tests Action"
        git status
        git remote -v
        echo See https://github.com/rusefi/rusefi/issues/2446
        git pull https://github.com/rusefi/rusefi master
        git add unit_tests/triggers.txt
        git status
```

rusefillc, 2021-07-21:
looks like we have two different actions generating configs

we have the official generate configs action

and we have unit tests which want to commit triggers which also generate config

What tools do we have? Clearly those two jobs do not want to be parallel. Maybe we should not try to run those in parallel? Maybe it's time to either learn how to chain actions, or simply merge those two actions together?

chuckwagoncomputing, 2021-07-21:
Has anyone here used git hooks? Is it possible to have git hooks that automatically work for whoever is committing to their fork, without them manually setting them up?
They'd still have to have the necessary software on their computers I guess.

Having workflows that commit to the repo has been trouble from day one.

rusefillc, 2021-07-21:
in the office we have git hooks to block some merges
in the same office we lock branch while we do something very similar to our process here: in the office we have a daily automated process which reads, commits and tags master. So for the duration of the process we lock branch allowing only one user (said automation) uncontested access

I am really starting to think that the most cost effective major improvement would be to have only ONE automation to push into master. As is we have two? three? different automations competing. Merging trigger push into config generator looks to be my preference at the moment

As an optimization for the single job changing automation I would suggest two pushes there:
keep existing config push as first push ASAP, and then take all the time we need to compile unit tests and push fresh triggers since trigger is a way less frequent operation

rusefillc, 2021-07-21:
Can we please:
1) remove trigger push from unit_test jobs
2) keep trigger push jobs
3) append trigger generator (i.e. unit tests) and trigger push at the end of existing gen_config job

mck1117, 2021-07-21:
I like that - exactly one build with a `git push`, and it does all the things that need to be pushed.

closed 2021-07-24 by rusefillc

chuckwagoncomputing, 2021-07-24:
What? How is this fixed?

rusefillc, 2021-07-24:
I believe that https://github.com/rusefi/rusefi/commit/f5acbdd5fb5cdde2e1573a42f310b06a270445f0 is helping it a lot

commit 9a360984 (2021-07-24): GHA reliability story triggers/config fix #3010
commit f5acbdd5 (2021-07-24): GHA reliability story triggers/config fix #3010