rusefi/rusefi #3011: GHA reliability: retry FTP upload

rusefillc, 2021-07-21:
We definitely have FTP upload reliability problem which affects less important TS plugin but also more important pinout generator and else

are there any cost efficient ways to improve this area? Does linux offer a build-in or popular retry tool? https://serverfault.com/questions/80862/bash-script-repeat-command-if-it-returns-an-error has a link to something on GH

chuckwagoncomputing, 2021-07-21:
Why is FTP failing?

rusefillc, 2021-07-21:
I believe that FTP is failing sporadically because my hosting company is not perfect in terms of FTP reliability

mck1117, 2021-07-21:
can you link an example failure build on gha?

rusefillc, 2021-07-21:
https://github.com/rusefi/rusefi/runs/3119536573?check_suite_focus=true

[image]

rusefillc, 2021-07-21:
oh did not realize there are retries! not sure if there is a delay between those

chuckwagoncomputing, 2021-07-22:
I think the delay is pretty short, like <5 seconds.

rusefillc, 2021-10-09:
Related to https://github.com/rusefi/rusefi_documentation/issues/164

rusefillc, 2021-10-09:
this drama could be related to the increase in build configurations number

if GHA runs too many actions in parallel 1and1 could be considering this a bit of a DoS attach and limiting us.

rusefillc, 2021-10-09:
[image]

rusefillc, 2021-10-09:
Maybe progress?

[image]

I will keep an eye on this since I get email notifications on failed runs

rusefillc, 2021-10-09:
hold one where are my 999 retries? :(

[image]

chuckwagoncomputing, 2021-10-22:
You appear to be using an account without full SSH, only SFTP.

rusefillc, 2021-10-22:
So they are pushing us into bigger change. Please add path variable and lets go with universal account secret name

Primary account has full ssh

rusefillc, 2021-10-22:
RUSEFI_DOXYGEN_FTP_USER not removed yet
RUSEFI_DOXYGEN_SSH_PASS not removed yet

RUSEFI_SSH_USER added
RUSEFI_SSH_PASS added

rusefillc, 2021-11-07:
@chuckwagoncomputing looks like we've addresses all of these?

closed 2021-11-07 by rusefillc