rusefi/rusefi #3012: pinout documentation - anchor links to specific pins

rusefillc, 2021-07-21:
I would like to href/link pin 23 on the while connector of https://rusefi.com/docs/pinouts/proteus/

is it possible to have a URL specific to pin 23 of while on https://rusefi.com/docs/pinouts/proteus/ and any other pin?

chuckwagoncomputing, 2021-07-24:
Done

rusefillc, 2021-07-24:
@chuckwagoncomputing can you give me an example? what's a URL for white pin 23 for proteus?

chuckwagoncomputing, 2021-07-24:
That would be something like https://rusefi.com/docs/pinouts/proteus?connector=white35&pin=23
Depending on what is put in the `id` field in the yaml.

rusefillc, 2021-08-08:
As in - do we have any working links? :)

chuckwagoncomputing, 2021-08-09:
Yep, MRE main connector links are working.
Example:
https://rusefi.com/docs/pinouts/microrusefi/?connector=main&pin=22

rusefillc, 2021-08-10:
How do I click and on a pin and copy URL for that pin to share it online? i.e. how does people discover this feature?

chuckwagoncomputing, 2021-08-10:
If you click on a pin, its URL will appear in the URL bar of your browser.

rusefillc, 2021-10-02:
@chuckwagoncomputing I am still a bit confused :(

1) https://rusefi.com/docs/pinouts/microrusefi/
2) I click on yellow pin 1 
3) that gives me https://rusefi.com/docs/pinouts/microrusefi/?connector=main&pin=1
4) I close browser
5) i open https://rusefi.com/docs/pinouts/microrusefi/?connector=main&pin=1

expected result: some sort of scroll/highlight of pin 1
actual result https://rusefi.com/docs/pinouts/microrusefi/?connector=main&pin=1 looks same as https://rusefi.com/docs/pinouts/microrusefi/