rusefi/rusefi #3014: Let's admit it it's time to move to gradle

rusefillc, 2021-07-21:
Historically all java code was build by ant

At the moment we already have android app building using gradle so much of the code already builds under gradle and we have both and and gradle which is not optimal

we shall completely migrate from ant to gradle for all java code

rusefillc, 2021-07-24:
Related to #2857

rusefillc, 2021-07-30:
@Come1LLF00 is this related to this change? two sets of generated files looks like one set uses wrong path?

[image]

rusefillc, 2021-11-05:
this could be a great task to offshore with $$$ once I have $ and time to manage

rusefillc, 2022-11-19:
overall status: we are pretty close!

very few outstanding issues:

open question java console artifact size if compiled with ant VS if compiled with shadow_jar gradle task

open question of java console shadow_jar has all .dll and other resources

rusefillc, 2022-11-21:
@nmschulte we already have java_console gradle it just PROBABLY does not produce amazing result

```
cd java_tools
gradlew :ui:shadowJar
```

nmschulte, 2022-11-21:
(OpenJDK 8) `ant jar`: 11249099 bytes - 100%

```
rusefi/java_console$ ls -l ../java_console_binary/rusefi_console.jar 
-rw-r--r-- 1 nmschulte nmschulte 11249099 Nov 20 21:28 ../java_console_binary/rusefi_console.jar
```

(OpenJDK 8) `./gradlew :ui:shadowJar`: 23177023 bytes - 206%
```
@desmas-l-hpdo:rusefi/rusefi/java_tools$ ls -l ../java_console/ui/build/libs/ui-all.jar 
-rw-r--r-- 1 nmschulte nmschulte 23177023 Nov 20 22:35 ../java_console/ui/build/libs/ui-all.jar
```

nmschulte, 2022-11-21:
Here's a repository w/ the contents of the `gradle` jar atop the `ant` jar; the last commit shows a lot of extra files, many of them `.../test/` files, though the `ant` jar has some of this as well.

https://github.com/nmschulte/rusefi-java_console-jar-ant-gradle-diff

rusefillc, 2022-11-21:
@nmschulte thank you this was extremely helpful I did not realize close we are!

without com.ibm.icu jar file is only 20% larger I can live with that. Next step understanding the differences around embedded .cpp source code resources and checking what files does ant version have which are not present in gradle version

nmschulte, 2022-11-21:
This patch removes the `com/ibm` dir but bloats the ZIP to over 9MB!  The actual contents unzipped are smaller though, so I'm not sure what's at play.

```
diff --git a/java_console/ui/build.gradle b/java_console/ui/build.gradle
index c22363e02f..1f28ce01ec 100644
--- a/java_console/ui/build.gradle
+++ b/java_console/ui/build.gradle
@@ -28,4 +28,8 @@ shadowJar {
                 'Main-Class': 'com.rusefi.Launcher'
         )
     }
-}
\ No newline at end of file
+
+    dependencies {
+        exclude(dependency('com.ibm.icu:icu4j:.*'))
+    }
+}
```

Ant jar, `du -acs` of contents:
```
24729992	total
```

Gradle jar w/ above patch, `du -acs` of contents:
```
21523590	total
```

rusefillc, 2022-11-21:
Bloats? 

11249099  is 11M right?
23177023 is 22M right?
the patch bloats from 22M to 9M does it?

nmschulte, 2022-11-21:
aha .. :blush:

commit fa35138c (2023-07-04): only:Let's admit it's time to move to gradle fix #3014
commit adc0a086 (2023-07-04): only:Let's admit it's time to move to gradle fix #3014
closed 2023-07-04 by rusefillc