rusefi/rusefi #3015: sync nb2 vvt faster

mck1117, 2021-07-21:
In one log, we sync the crank after only 4 teeth, but VVT takes an additional 1.63 seconds to sync.  The culprit is probably:

```
	if (cisnan(oneDegreeUs)) {
		// todo: this code branch is slowing NB2 cranking since we require RPM sync for VVT sync!
		// todo: smarter code
		//
		// we are here if we are getting VVT position signals while engine is not running
		// for example if crank position sensor is broken :)
		return;
	}
```

mck1117, 2021-07-21:
moving that check after the vvt decode achieves VVT sync a full 1.3 seconds faster

rusefillc, 2021-07-23:
Relates to #3024
See first attempt at #3013

mck1117, 2021-07-23:
minor clarification, #3024 and #3013 are about _correctness_ not _fast sync_

rusefillc, 2021-07-23:
_Extemely_ vaguely related by means of same test mule etc. Just need any ticket to tie multiple PRs together :)

mck1117, 2022-02-16:
This is pretty good for now.

closed 2022-02-16 by mck1117