rusefi/rusefi #3030: "prettyClose": some strange code inside idle controller?

andreika-git, 2021-07-22:
1) We have IdleController::update() where we call:
```
float position = getIdlePosition();
applyIACposition(position PASS_ENGINE_PARAMETER_SUFFIX);
```

2) Inside IdleController::getIdlePosition() we set:
```
engine->engineState.idle.currentIdlePosition = iacPosition;
return iacPosition;

```

3) Inside applyIACposition() we check:
```
bool prettyClose = absF(position - engine->engineState.idle.currentIdlePosition) < idlePositionSensitivityThreshold;
if (prettyClose) {
	return;
}
```

Does it really work? 'prettyClose' equals true all the time, right? I cannot get past it on the bench.

mck1117, 2021-07-22:
I've always used `idlePositionSensitivityThreshold` = 0, so I've never hit this.  What is the purpose of idlePositionSensitivityThreshold?

andreika-git, 2021-07-22:
For stepper motors, it equals to 1 step to avoid excessive micro-movement.

rusefillc, 2021-07-22:
```
idlePositionSensitivityThreshold = 1.0f / engineConfiguration->idleStepperTotalSteps;
```
how does it equal "1"?

stepper is already mention around there, I've made it mentioned twice

mck1117, 2021-07-22:
it's not equal to 1, it's equal to one stepper step worth

mck1117, 2021-07-22:
wait - I'm not sure we need `idlePositionSensitivityThreshold` at all - the stepper driver truncates to the nearest step, so sub-step moves don't trigger a move anyway.  With `idlePositionSensitivityThreshold` set to 1/stepCount, it will just prevent sub-step moves twice.

andreika-git, 2021-07-23:
`"the stepper driver truncates to the nearest step, so sub-step moves don't trigger a move anyway"`

If a value is right in the middle between two steps, then the stepper still moves! For example, it's 10.99, and then becomes 11.01. The small 0.02 increment will trigger stepper to move 1 step because of truncating to nearest (the same would be with rounding too).

This 'idlePositionSensitivityThreshold' appeared not in an abstract world, it was caused by a real car experience with a low-step-count stepper, and was tested well enough to see that the IAC works **better** with it.

The current implementation has two issues:
1) position == currentIdlePosition is always true, and the stepper just doesn't work
2) position is in percentage, and idlePositionSensitivityThreshold is ratio

mck1117, 2021-07-23:
Ah, that makes some more sense then.

PR #3038 (2021-08-02): perform extra-step-prevention in the stepper driver itself
`idle_hardware.cpp` shouldn't care about stepper-specific problems, so do that in the stepper driver itself.  Don't update setpoint unless we're at least one full step away from the last setpoint.

fix #3030
closed 2021-08-02 by rusefillc