rusefi/rusefi #3033: Implement auto knock threshold sampling

OrchardPerformance, 2021-07-22:
After a load of data gathering and analysis it has been shown a delayed sampling window positioned just after combustion is able to sample a pretty good knock baseline without end user interference. 
Test was performed with the phasing 120 degrees out from normal sampling but I suggest we give it a go with something like 45 degrees out and a shorter sampling window. 

My proposal is that we set something up so that the ECU takes a knock baseline value out of time with the combustion once every 10 (maybe?) cycles and that is spat out into the log to be able to fill out the baseline table.
It could also be used to adapt the baseline table over time if we had some kind of keep alive memory.
Alternatively we could click the ECU into "baseline finder" mode and have every sample taken out of phase for a short time and then manually turned into a table.

TLDR: Implement an adjustable offset that we can use to take a knock baseline and dump into a log.

mmatousek89, 2021-07-23:
theese baselines should be diferent for difert boost targets ie. diferent manifold absolute pressure values, not only rpm plays its role here

rusefillc, 2021-08-10:
@mck1117 @OrchardPerformance  two changes
1) what was hard-coded as 45 is now configurable
[image]

2) additional new behavior added: at the end of the window last raw and filtered voltage would be assigned into debug mode "knock" gauges

OrchardPerformance, 2021-08-10:
We need to be able to retard (move later after TDC) that 45 degree window as well. 
It needs a box to put in a value that adds crank degrees on to the start and finish of the window.

rusefillc, 2021-08-10:
Once my latest commit builds in 20 minutes from now we would have knockDetectionWindowStart as an angle between spark and start of sampling

rusefillc, 2022-04-29:
@OrchardPerformance what's the state of this issue?

OrchardPerformance, 2022-04-29:
Probably dead. 
Out of phase sampling only seems to work on some engines, mostly those with 120 degree cranks (3 and 6 pot).

We can probably still do an auto measurement when map is low but I'm not sure.

We need to think of something as the only option I can offer right now is to gather the data and post process in Matlab. 
Good luck with teaching users to do that. 😂