rusefi/rusefi #3044: Clutch switch inverted options do not invert detected pedal position

Keith-N, 2021-07-24:
Using Hellen 64 Rev A with 1991 Miata. The clutch up switch input functions properly by default, the clutch down switch input is inverted and detects the pedal as down while it is up.  Attempting to use any of the clutch up/down switch inverted options does not change the behavior of either input.

rusefillc, 2021-07-24:
please paste a URL to your tune here

once I have the URL we will try "readpin XX" command in rusEFI console to gather a bit more data

rusefillc, 2021-07-24:
@Keith-N I mostly do NOT understand what this ticket is about

can you please spell out expected behavior with twice more words? can you please spell out actual behavour?

@BeerMoneyMotorsports can you help with translation here?

BeerMoneyMotorsports, 2021-07-24:
@rusefillc the input for clutch switch is flipped. When pressed, TS shows clutch up. When released, it shows clutch down. So The ecu is showing the opposite of what’s happening.

Keith-N, 2021-07-24:
Tune file here: [https://rusefi.com/online/view.php?msq=601](url)

The input for clutch down shows the pedal as pressed while it is not, and the input for clutch up is okay. The inverted options do nothing or do not affect the detected clutch pedal position.

When the clutch is up I get 
[image]

When the clutch is pressed I get
[image]

Using the inverted options in the sensor inputs menu does nothing to change the behavior no matter what is selected. This has been tested with the clutch set to only a single input and all of the inverted options tested with no change.
[image]

This also impacts launch control and causes it to be enabled when the clutch is up instead of when it is pressed regardless of the inverted option selected.

rusefillc, 2021-07-24:
https://rusefi.com/online/view.php?msq=601&dialog=otherSensorInputs shows invalid configuration with same pin assigned to two different functions

I've just improved tooltips to make things less confusing, please figure out which of the two different switches do you have

these specific settings do not apply on the fly
I've just change project so that it would be a bit more obvious that ECU reboot is required once these settings are changed

as for actual code in case of valid config with valid reboot after settings change I do not see a problem yet but I see weird 
```
! do we need these? why does pin mode not do the job?
	bit todoClutchUpPinInverted
	bit todoClutchDownPinInverted

```

so we seem to have 3 or 4 unrelated issues here

Keith-N, 2021-07-24:
The invalid configuration must have been set from when I was playing with the settings.

rusefillc, 2021-07-25:
@Keith-N @BeerMoneyMotorsports please test again with latest firmware and latest project

we now have explicit "invert" fields in addition to "mode" fields, and changes should now be applied on the fly

Keith-N, 2021-07-25:
After testing with the new firmware, the invert option does properly work and changes are able to be done on the fly. 
Using TS and observing the clutch status indicators, both the clutch up and down inputs were tested separately. By default for the Miata, the pedal down status would show the pedal as down while it was up. Setting invert to true would correct this and show the pedal as down while it was pressed. The invert setting also worked with the clutch up input. This also corrects the behavior of launch control with the clutch down input being properly detected.

BeerMoneyMotorsports, 2021-07-25:
@Keith-N tldr: it works?

Keith-N, 2021-07-25:
@BeerMoneyMotorsports Correct, it is working for me.

closed 2021-07-25 by mck1117

BeerMoneyMotorsports, 2021-07-25:
@Keith-N thankyou! I’ll test tomorrow