rusefi/rusefi #3045: 2 byte table axis

rusefillc, 2021-07-24:
at the moment we are flexible with TS table cell size - it could be 1 byte or 4 byte

but each axis cell still always take 4 bytes. for 8x8x1 table we have 8x2x4 bytes for axis would be nice to have an option of axis taking less memory than data in the table

step 1 extra flexibility for 3D interpolation template?
step 2 something else?

rusefillc, 2021-08-05:
should it actually be 1 byte for RPM?

how do we want to pack load axis into two bytes?

mck1117, 2021-08-10:
> how do we want to pack load axis into two bytes?

we already do it in a few places, scaled by (iirc) 50 or 100

rusefillc, 2021-12-06:
@alxrMironovHome can you help?

rusefillc, 2021-12-21:
Relates to #3707

rusefillc, 2022-01-12:
@mck1117 shall we close?

closed 2022-01-12 by mck1117