rusefi/rusefi #3048: Rename "2 wire wasted" in TS ignition settings

MHTSOSVOODOO, 2021-07-24:
We should rename the field "2 wire wasted" in TS ignition settings because this question come up all the time in support slack. New users not understand what it means and most don't see the pop up explanation. 
"Wasted spark with individually wired coils" should be clear enough for everyone.

rusefillc, 2021-07-24:
Thank you for a great improvement proposal!

We do not have "2 wire wasted". You are asking me to do the research and figure out that what you MEAN is "Two wire"

Please help me help you by providing as much clarify and details on the tickets as possible. Sometimes a screenshot is just 4 clicks for you and a lot of help for me.

rusefillc, 2021-07-24:
@MHTSOSVOODOO done should be ready to test in 29 minutes from now

commit 2cc02720 (2021-07-24): Rename "2 wire wasted" in TS ignition settings fix #3048
commit c4dac4ed (2021-07-24): Rename "2 wire wasted" in TS ignition settings fix #3048
closed 2021-07-24 by rusefillc