rusefi/rusefi #3054: Usability: unify MAP, AFR and CLT/IAT approach

rusefillc, 2021-07-24:
at the moment we have three types of sensors handles in three different ways

MAP and AFR have presets but those are handled differently

CLT/IAT does not have named presets

we probably need a higher level of unification

rusefillc, 2022-01-23:
@BeerMoneyMotorsports any chance you would be interested to attempt the settingSelector for MAP sensors?

BeerMoneyMotorsports, 2022-01-23:
With guidance, absolutely. This will be a lesson in how to configure base settings?  Or  get me started in tweaking base tunes and presets?

rusefillc, 2022-01-23:
@BeerMoneyMotorsports https://github.com/rusefi/rusefi/pull/3326/files is the guidance

nope that would have nothing to do with per-vehicle base tunes :(

BeerMoneyMotorsports, 2022-01-23:
Fuck it, ill give it a try.

mck1117, 2022-01-23:
Base tune settings will have to change to setting the values (voltage/pressure) instead of a preset, if that's what they're doing today.  The goal is to remove the presets from the firmware, and do it completely in TS instead.

BeerMoneyMotorsports, 2022-01-24:
looking over it, i will need some tutelage to accomplish this.  It may not be optimal for me to attempt.