rusefi/rusefi #3063: pedal logic

Abricosvw, 2021-07-26:
{ throttlePedalPositionAdcChannel != @@ADC_CHANNEL_NONE@@ || (useStepperIdle && useHbridges) }
7:01
set a pedal
i  think these settings should be open 

[image]

rusefillc, 2021-07-26:
@mck1117 funny case:

{ useStepperIdle && useHbridges } is one way to use BOTH H-bridges, we do not disable etbFunctions[2] in case of { useStepperIdle && useHbridges }

what is the order of preference between { useStepperIdle && useHbridges } and say wastegate?

Abricosvw, 2021-07-26:
i have H-bridge wastegate (DC motor potenciometro feedback  )
i have H-bridge idle ( DC motor  potenciometro feedback  ) 

by the way a guy from Europe  with Honda turbo motors for buggy car have same H-bridge wastegate