rusefi/rusefi #3070: Faster enginePins.stopPins()

rusefillc, 2021-07-28:
at the moment we can half coils/injectors stuck open for up to half a second on loss of trigger input. in reality that usually comes with relays going off but still maybe we shall detect this condition faster for any trigger or at least consider trigger tooth count in the delay calculation

```
	if (secondsSinceTriggerEvent < 0.5f) {
		// Engine moved recently, no need to safe pins.
		return;
	}

```

rusefillc, 2021-07-28:
@mck1117 please upload video if feels appropriate

rusefillc, 2021-08-19:
```
Alex Miculescu  9:45 AM
I was doing timing measurements and left ign on and killed tge engine from ts

9:45
3rd start it was running on 2 cyl
9:47
So probably that killed it
9:47
Because when it runs, it is cool and working
```

TODO @rusefillc @mck1117 to check how is that specific scenario handled

mck1117, 2021-08-19:
I feel like the engine stop button should work via limp manager, instead of doing whatever crazy logic (force killing the pins??) it has now

mck1117, 2021-11-29:
will be fixed by #3180

mck1117, 2022-02-16:
This is fixed both by better engine stop logic, and by overdwell protection.

PR #3077 (2021-07-30): instant rpm uses full engine cycle
fix #3070 

The problem was that instant RPM was trying to use `currentCycle.current_index`, which doesn't account for trigger patterns that only cover part of the engine cycle (for example, a crank only trigger).

This looks much better now!

[image]

As a bonus, this probably helps #3015, since now symmetrical crank instant RPM /actually/ works, instead of only calculating on one tooth per cycle, which effectively broke instant RPM on symmetrical crank triggers.
closed 2022-02-16 by mck1117