rusefi/rusefi #3074: Device goes offline/re-enumerated after timeout

dron0gus, 2021-07-29:
Warning: custom HW - Subaru EG33 board.

First High Speed logger stops.
Then
[image]
Then
[image]

And after some timeout linux resets and re-enumerate RusEFI device.

[13141.443038] usb 2-1.8.3.1.3: reset full-speed USB device number 18 using ehci-pci
[13141.652099] cdc_acm 2-1.8.3.1.3:1.1: ttyACM2: USB ACM device
[13141.661839] VFS: busy inodes on changed media or resized disk sdd
[13141.662700] sd 7:0:0:0: [sdd] 2048 512-byte logical blocks: (1.05 MB/1.00 MiB)

After it is possible to connect to device with TS over new ttyACM.

rusefillc, 2022-11-14:
@dron0gus is this still an issue?
@dron0gus do you have time and motivation to test this on MRE or Proteus?
@mck1117 do you have pure bare metal linux to try? have you experienced that or similar issues?

dron0gus, 2022-11-15:
I'll check if it is still reproducible.
May be related to https://github.com/rusefi/rusefi/issues/4359 . But this should not cause re-enumeration.