rusefi/rusefi #3114: dual / multiple fuel maps

rusefillc, 2021-08-05:
popular question is dual / multiple fuel maps

Владимир 5634 from whatsapp elaborates that this is about multiple boost modes in his specific case

rusefillc, 2021-08-05:
Depends on #3045

mck1117, 2021-08-10:
obligatory X-Y problem question: what are users expecting to change at runtime that will actually change the fueling such that it needs a full separate VE table?

rusefillc, 2021-08-11:
Pavel says he needs two maps:
1) eco mode, up to 4k highway driving
2) sport mode

NA engines

mck1117, 2021-08-11:
But why would the *VE MAP* be different? What does he actually want to change? There should be no change to the VE map based on what the driver wants.

His desire sounds like he might actually want to *ETB pedal maps*, not two *VE maps*

shadowm60, 2021-08-13:
In eco you need higher afr, you want less fuel consumtion, in sport you want different target afr. I thing there should be two target afr tables, two ve and two ignition maps. Based on a switch select the mode?

mck1117, 2021-08-13:
I disagree.  Your target AFR table should be shaped so that you're in an efficient regime while cruising, and a best-torque regime while floored.  Ignition is even easier: everywhere should use the same ignition timing.  Ideal power and ideal efficiency are *identical* for ignition.  There certainly shouldn't be two VE tables - as that's a fundamental constant of the engine, not anything to do with how you want the car to drive.

I doubt any OEM actually changes anything other than pedal->tps relationship and boost target in different modes. Target AFR, timing, and VE have no reason to change.

AlexHLinS, 2021-08-14:
The different VE map needed when VE dramatically changed by smth, for example vtec on/off, Vvti etc. Two maps sport and eco will realize only by accel pedal reaction or target boost (but we talking about NA). Also changing ignition map, afr map etc is not an solution, coz best mpg <-> max torq

dron0gus, 2021-09-03:
We need separate limp mode VE table (TPS/RPM).
I think it will not look exactly the same as MAP VE table?

We (at least in Russia) need few AFR/advance tables due to absence of good/sport fuel on each gas station.

rusefillc, 2021-09-04:
Simon - Orchardperformance  7:25 PM
I agree with your comments. Correct tuning for mbt is correct for mbt and stay out of enrichment ranges.
Its something that comes up all the time.
Doing something to suit everyone...
Back to the main map + modification table idea, if we make the reason for switching to the modification table user configurable then people can set it to what ever they want.
The become just condition A for main table, condition B for 100% modifiction table.
Doesn't matter if condition B is eco mod, NOS, high boost, race gas or mother in law mode.

7:26
Honestly I think just adding maybe 4 selectable 8x8 modification tables and using them for VVT, eth, map switching, etc is the best catch all that kills a lot of targets in one go
7:28
On the other side, cam timing changes in eco mode but its not a VE map change its a total engine operating point change. Like for like the VE at that operating condition is still the same regardless of mode.

Matthew Kennedy  7:43 PM
There might be technical challenges with that owing to the fact that we use tunerstudio
7:43
Link ecus have some memory region for table storage, and you can configure which function uses which tables
7:44
But the software knows about that
7:44
There's no way, afaik, to tell TS to switch WHERE a table is located based on another setting

Simon - Orchardperformance  8:07 PM
Hmmm
8:11
How does FSIO handle that curently?
Is there something that acts as a place holder within the various calculations, so instead of having to point to a table we point the table to the place holder?

Dave B.  1:47 AM
Tunerstudio... 90% awesome

Matthew Kennedy  1:48 AM
@Simon - Orchardperformance Fsio doesn't handle that

Simon - Orchardperformance  8:11 AM
Just trying to understand the nature of the issue, but does tuner studio need to know about it at all?

Simon - Orchardperformance  8:18 AM
I guess my point is, do the tables need to switch where it is located?
If we take a situation where we were going to add 2x 16x16 Timing and Fuel tables like an old dual table ECU, if those tables were broken into 8x 8x8 tables then we could allocate 2 as a AFR modifier, 2 as a VE modifier and 4 as a timing modifier for the same amount of RAM?
If we put some "hooks" into the calculations we would want to modify with them and set the default action of those hooks to be "no modification" then surely all we would need TS to do is to configure if that hook is doing nothing or bringing in the modification.
Think a drop down with "no modification", "+1 dimension", "+2 dimensions".
8:20
If the issue is about being able to choose the trigger of the modification then I would have thought existing systems would be able to do that, If each table has a 2d "bias" table, like the one we were saying is a % to % style mapping, then is it not simply a case of switching the axis ID for that table?

Simon - Orchardperformance  8:26 AM
Say if we want to blend from the main timing table to "timing modifier 1" based on flex fuel sensor then we set in TS the setting for "+1 dimension" which activates the calculation that adds the value of the modification table * bias table to the spark demand.
We then set "bias table 1" axis to be flex fuel and that allows 0->100% of the additional "modification table 1"
8:28
I know that idea limits us to only a couple of VE, AFR and Spark modification but even then, wielded well I cannot see how someone would want to try tuning more than 6 dimensions of fuel modification, it would be a mess.
8:30
I guess my TLDR there is "do we need to switch anything, just simply activate or ignore its affect on the calculation"

Simon - Orchardperformance  8:38 AM
I am also happy to accept that I am completely misunderstanding the problem statement

BeerMoneyMotorsports  11:29 AM
image.png 
image.png

11:29
do we really need both?

Simon - Orchardperformance  2:03 PM
Probably yes
2:04
Although they do similar things one is MUCH simpler than the other

Matthew Kennedy  2:04 PM
we need something TPS based
[... 195 lines cut ...]
seems to me that 'modifier' is way more complicated.  'switch' means change the address.  'modifier' means add an extra address and change the algorithm.  of course, it's just software.  :slightly_smiling_face:

Zach Hoobler  7:46 PM
I see modifier as "add these tables together when modifier enabled"
7:47
not really all that different from "replace this table with that one"

Matthew Kennedy  7:47 PM
then you have to support blending
7:47
base + (x1 * table1) + (x2 * table2) + ... + (xn * tablen)
7:47
which is ok
7:50
@Dave B. that give you any flashbacks to ford mapped point vvt shit?
:eyes:
1

Simon - Orchardperformance  8:00 PM
Supporting blending will be needed anyway
8:00
At least for handling the cam transitions and multi fuel

Matthew Kennedy  8:00 PM
for flex at least, yes

rusefillc, 2021-11-30:
@NMSTEC is asking for dual VE and timing table since front and real cylinders on HD behave differently due to V-twin and cooling and whoknows what else

AlexHLinS, 2021-11-30:
Could we fix it by individual cylinder trim?

mck1117, 2021-11-30:
We're conflating two different problems/features here.

1. Switchable/blendable maps
2. Per-cylinder trim maps

BeerMoneyMotorsports, 2022-01-12:
Can we close this and create a ticket that adds timing, cranking and etc tables?  Flex is a feature that’s becoming more common.  Fuel blending the ve table is fine but there are cranking and ignition table(that i know) are different between fuels.

NMSTEC, 2022-01-12:
This is now a bit of a mess. This can also be taken as Dual Maps as I need them for enrichment. I'd say "User selectable performance maps" would be a better new ticket.

rusefillc, 2022-01-12:
@NMSTEC please explain WHY and WHAT performance means separate maps?

NMSTEC, 2022-01-30:
Multiple uses for this, and reasons.

1. Limp mode (Harley will need this, if meth runs out, overboost, excessive knock, CLT too high)
2. Different fuel map (Instead of running 91 & meth, when no available, select low octane map with retarded timing, and more enrichment)
3. Scramble mode (When you want much more power than normal, for a short period of time)
4. Different fuel map again, but this time for E85. It has massively different everything in regards to timing and fuel.

mck1117, 2022-01-30:
How many of those cannot be done in lua?

1. Limp mode: subtract 0.2 from target lambda, subtract 5-10 degrees from timing
2. See 1
3. See 1, but do the opposite
4. E85 support is a somewhat separate problem, we already do automatic mixture adjustment for E85, and your target lambda will at least be close to the same as on gas. E85 timing blend is #3778 
5. Not a good reason!

NMSTEC, 2022-01-30:
1. How would you do this in LUA without as you say, eating too much cpu time? Especially on the budget F4 HD is going to be running.
2. Fair enough.
3. This is stretching it.
4. -
5. Is good reason. What's a good reason for not doing this?

duhafnusa4, 2022-02-28:
this needs updates

rusefillc, 2022-06-22:
Josh Stewart says

We had a similar conversation with Speeduino. The most common arguments were around things like being able to switch to a different fuel (Eg LPG) without needing to manually change the tune and being able to do blended modes for setups such as boosted ITBs.
The dual tables with blending works really well for the latter now, so it ended up being a nice addition.

biohazardav, 2022-08-22:
I don't understand why it's such a big argument to add switchable or blended fuel and spark tables. Many other engine management solutions offer this and is required for things like variable valve timing, VTEC and cammed/boosted engines with other switchable power adders. What would it take to make this happen?

rusefillc, 2025-02-26:
hello from 2025!

blends are the way.