rusefi/rusefi #3117: funny business trigger wheel correlation with cylinder count

rusefillc, 2021-08-08:
is all this about quick cranking? what's the quick cranking strategy they use?

Nissan 360 CAS number of windows matches cylinder count

v6 nissan is 
[image]

i4 nissan is
[image]

rusefillc, 2021-08-08:
See #3118
See v6 picture for reference https://rusefi.com/forum/viewtopic.php?p=41453#p41453

mck1117, 2022-02-16:
Yes, nissan used a sort of goofy cam pattern that happens to blip out the next cylinder's number.

Is there new code to write for this?