rusefi/rusefi #3121: analog input for trigger offset

mmatousek89, 2021-08-09:
potentiometer connected to analog input 

coarse setup 0V = - 720deg  5V= +720deg
fine setup 0V = - 72deg 5v = +72deg

help with setting up trigger offset

rusefillc, 2021-08-09:
https://xyproblem.info/ interesting request but really need to know what problem are we solving

mmatousek89, 2021-08-09:
thts not problem, thats feature to improove workflow, you can put it under "nice to have" label

mck1117, 2021-08-10:
So why not just use +- buttons in software? This is big X-Y problem.  You're going to have to convince us that it improves the workflow.  Setting trigger offset is something that users spend maybe 3 minutes on total, ever.  It's just not worth it for somebody to wire up an additional knob to use for 3 minutes of trigger offset config. It's just not that hard a problem to do it with a pc.