rusefi/rusefi #3122: analog output for crank angle

mmatousek89, 2021-08-09:
fast DAC providing 0-5V coresponding to 
0-720deg crank angle

probably special hardware?

can we update every crank degree or it will owerload processor?

this can be handy for osciloscope in X/Y mode or fast datalloger to have synchronisation with crankshaft without dificulty of triggerpattern decoding

rusefillc, 2021-08-09:
https://xyproblem.info/ what problem are solving

mmatousek89, 2021-08-09:
thts not problem, thats feature to improove workflow, you can put it under "nice to have" label

mck1117, 2021-08-10:
but why? why not just sync the scope to an ignition output or something?

We don't have the hardware to be able to do this.  We can't reasonably do this without crazy CPU load.

closed 2021-11-29 by mck1117