rusefi/rusefi #3128: rusefi console - troubles with 'online'

thirstyone, 2021-08-10:
I get this error at my first attempt of using 'online' section (copy-pasted the code from the forum, hit 'Upload' button)
On my next attempts nothing happens at all. 

java.util.MissingResourceException: Can't find bundle for base name javax.xml.bind.Messages, locale ru_RU
	at java.base/java.util.ResourceBundle.throwMissingResourceException(ResourceBundle.java:2055)
	at java.base/java.util.ResourceBundle.getBundleImpl(ResourceBundle.java:1689)
	at java.base/java.util.ResourceBundle.getBundleImpl(ResourceBundle.java:1593)
	at java.base/java.util.ResourceBundle.getBundleImpl(ResourceBundle.java:1556)
	at java.base/java.util.ResourceBundle.getBundle(ResourceBundle.java:857)
	at javax.xml.bind.Messages.format(Messages.java:71)
	at javax.xml.bind.Messages.format(Messages.java:52)
	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:278)
	at javax.xml.bind.ContextFinder.find(ContextFinder.java:421)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:721)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:662)
	at com.rusefi.xml.XmlUtil.writeXml(XmlUtil.java:18)
	at com.rusefi.tune.xml.Msq.writeXmlFile(Msq.java:78)
	at com.rusefi.tools.online.Online.uploadTune(Online.java:81)
	at com.rusefi.ui.OnlineTab$1.actionPerformed(OnlineTab.java:30)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6632)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6397)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5008)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

rusefillc, 2021-08-10:
@thirstyone is that a Win or a Mac? What version of java are you using?

I bet that's some drama related to JAXB removal from java 11 or something similar

thirstyone, 2021-08-10:
java -version
java version "11.0.2" 2019-01-15 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.2+9-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.2+9-LTS, mixed mode)

thirstyone, 2021-08-10:
yep, osx

rusefillc, 2021-11-05:
kind of related to #1526

rusefillc, 2021-12-22:
@alxrMironovHome that could be a trivial fix :( but someone has to try with adding of jaxb into classpath.

rusefillc, 2023-07-03:
java 8 is no longer suppored

closed 2023-07-03 by rusefillc