rusefi/rusefi #3141: AbricosVW still has issue on Burn

rusefillc, 2021-08-12:
@Abricosvw reports that he still have an issue on burn

Maybe something FSIO similar to 3140? Maybe something else is re-initializing?

I guess we need better reinit logging in console and a way to hit "burn" in console? "writeconfig" is not exactly burn, I assume we have more real "burn" just do not remember command name

rusefillc, 2021-08-12:
and we have second report #3144

rusefillc, 2021-08-14:
maybe https://rusefi.com/online/view.php?msq=612

abelom, 2021-08-15:
https://www.youtube.com/watch?v=U67EWOa0IpE

rusefillc, 2021-08-15:
We have engineSnifferRpmThreshold I will now have same limit applied to tooth logger

rusefillc, 2021-08-15:
Related to #3161

Abricosvw, 2021-08-17:
https://rusefi.com/online/view.php?log=315

[image]

rusefillc, 2021-08-17:
TPS drama was noticed at https://www.youtube.com/watch?v=U67EWOa0IpE&t=554s

TPS was replaced. https://github.com/rusefi/rusefi/issues/3166 created