rusefi/rusefi #3162: SD-card log fields improvement (warning counter & last code)

andreika-git, 2021-08-16:
I think we need to improve the list of SD-card log fields.
Currently we do not have any "warning counter" or "warning last code" in the logs, and they are essential for effective problem tracking.
Maybe we should also record the debug fields and debug mode?
Also we don't have engine CRC16 and tune CRC16 which prohibits from using the recorded SD-card logs effectively in REO.

rusefillc, 2021-08-16:
ideally all three kinds of logs should have same exact info

ideally we shall implement continues integration which counts log fields of all three approaches and compares those three numbers

rusefillc, 2021-08-17:
Just added "uptime" field into TS log - I believe that one would be useful for SD logs as well

commit d4a25c48 (2021-09-09): SD-card log fields improvement (warning counter & last code) fix #3162
closed 2021-09-09 by rusefillc