rusefi/rusefi #3167: TPS/TPS enrichment should be converted to grams or else

rusefillc, 2021-08-17:
We have converted starting fuel from ms to gramms

Should we do same for TPS / TPS table?

[image]

OrchardPerformance, 2021-09-21:
Should probably be a multiplier and just 1.00 = 100% (no change) 1.2=120%.
Reason is this adds onto the running fuel so should probably be proportional to the running fuel. 

If it is proportional then we are more likely to be able to set a good default basemap.

highspeed427, 2024-06-05:
Can we revisit this? I am running into major issues with this being ms. i agree with @OrchardPerformance on this matter. it needs to be a multiplier. Ive recently been building tunes for multiple difference size injectors and keep having to completely remap the accel fuel. 

thanks

closed 2024-06-09 by rusefillc