rusefi/rusefi #3181: Default 'primary' bundle should be compiled with hip9011 support

rusefillc, 2021-08-20:
Pavel Xenon is asking for subj

rusefillc, 2021-08-20:
Maybe depends on #3182

mi-hol, 2023-04-03:
@rusefillc as there is no longer a 'primary' bundle, I'd suggest to close

mi-hol, 2023-04-03:
additionally "We believe that we have hip9011 hardware yet nobody has ever actually detected knock with a hip9011, someone would have to figure our calibration approach and test the overall implementation." from https://github.com/rusefi/rusefi/wiki/knock-sensing would strongly contradict this request

rusefillc, 2023-04-03:
we no longer have default primary but still the same idea - the spirit of the request was for the "board agnostic" version of firmware to have hip9011. Exactly while we do not have proven results with it.

rusefillc, 2025-06-03:
hip9011 is gone

closed 2025-06-03 by rusefillc