rusefi/rusefi #3186: unclear SPI start error

rusefillc, 2021-08-23:
[image]

https://rusefi.com/online/view.php?msq=629

Denis says he is using nucleo144 stm32f429

rusefillc, 2021-08-23:
Trying to find if custom or not custom binary

Maybe specific SPI is not enabled on compile time?

rusefillc, 2021-08-23:
That's using https://rusefi.com/build_server/rusefi_bundle.zip

mck1117, 2021-08-23:
woohoo, that's our first error caught in the real world that would've caused an ECU lockup!

mck1117, 2021-08-23:
I suspect it's a DMA collision

mck1117, 2021-11-29:
We don't have the ability to change that string, as it comes directly from ChibiOS, not us.

rusefillc, 2021-11-29:
technically we can patch ChibiOS

mck1117, 2022-02-16:
I think that's the best we can realistically do in that scenario, it's essentially impossible that users hit this on any non-bare-stm32, so those messages are really for devs.

closed 2022-02-16 by mck1117