rusefi/rusefi #3187: Alex reports CUSTOM_RE_ADDING_INTO_EXECUTION_QUEUE = 9011

rusefillc, 2021-08-23:
https://rusefi.com/online/view.php?log=326
Alex says

What is strange i noticed RPM drop, and 9011 scheduler error code, that is described why it was logged, but i expected a trigger error or something

i do not know why my rpm dropped

nice log, you can clearly see when thermostat opened, and all the cold water from the radiator entered the engine :)))

rusefillc, 2021-08-23:
One thing I can think of is custom error code for each schedulingAdd

this case if we reproduce we would know at least which exact scheduling

@mck1117 thoughts?

shadowm60, 2021-08-23:
This happened in the morning only, i had to do a log with only 15 samples/sec since i have issues with usb diconnect, and maybe i have lost a warning message, but based on the warning counter, i did not. Please correct me if i am wrong.

I was expecting to see trigger issue, since this is how it felt, but i can only see this warning and from the log, you can clearly see the rpm drop at the start of the log.

mck1117, 2021-08-23:
[image]

there are also crazy vvt position spikes when those rpm drops happen, can you try disabling the cam input and see if it still happens?

mck1117, 2021-08-23:
There were no trigger errors, which supports the VVT theory

shadowm60, 2021-09-05:
After removing cam sensor, i cannot see this issue any more. I have configured cranck without cam sensor, but i do have a cam input configured, i had this issue, now with CAM sensor unplugged this is not visible any more.

The issue is caused by a cam trigger that is too short for the Hall sensor, originally this was VR as well, so i have to change the trigger notch to ensure a longer pulse, this way hall sensor will be stable. 

I think we can close this since it is not a FW issue.

mck1117, 2021-09-06:
this is in fact a duplicate of #3197