rusefi/rusefi #3191: 94/95 miata presets

BeerMoneyMotorsports, 2021-08-24:
Please adjust defaults below in 94/95 tune.

- injector size = 254cc
- engine displacement = 1.8

rusefillc, 2021-08-24:
I believe that we already have this
please clarify if that's not how the firmware acts

[image]

BeerMoneyMotorsports, 2021-08-24:
This is not how it functions for me.  
1- loaded latest nightly
2- Apply NA8 defaults
3- disconnect ECU 
4- Restart Tech Tool
5- Connect ECU

Below are screenshots of what i am seeing after applying the NA8 defaults.  Also defaults to Batch fire

[image]
[image]

rusefillc, 2021-08-24:
@BeerMoneyMotorsports I found at least one defect

let's try again in 20 minutes :)

rusefillc, 2021-09-02:
@BeerMoneyMotorsports I assume this was fixed

closed 2021-09-02 by rusefillc