rusefi/rusefi #3221: x86_64-w64-mingw32-g++: error: libsanitizer.spec: No such file or directory

rusefillc, 2021-09-05:
windows unit tests jenkins has failed

```
Compiling log_field.cpp
Compiling gtest-all.cpp
Compiling gmock-all.cpp
rm -rf build/obj/*gcda
Linking build/rusefi_test
x86_64-w64-mingw32-g++: error: libsanitizer.spec: No such file or directory
make: *** [../firmware/../unit_tests/rules.mk:170: build/rusefi_test] Error 1
/cygdrive/c/Program Files (x86)/Jenkins/workspace/unit_tests/unit_tests
```

rusefillc, 2021-09-05:
@mck1117 I had to partially rollback #3217 to save master

mck1117, 2021-09-07:
https://github.com/msys2/MINGW-packages/issues/3163

rusefillc, 2021-09-07:
@mck1117 do you want to make options conditional on NT / not NT in Makefile?

mck1117, 2021-09-07:
yes, we should just turn it off if NT

PR #3230 (2021-09-07): no asan on windows
fix #3221
closed 2021-09-07 by rusefillc