rusefi/rusefi #3251: Crazy MAP readings

OrchardPerformance, 2021-09-21:
User is reporting an issue where the MAP gauge and Y axis on the VE table do not agree. 

When free revving the engine (no load) the MAP gauge looks correct at max 100kpa but the VE table is spiking up to 250kpa. 

Professional opinion, boost is very unlikely to exist on free rev, suggests something is VERY wrong with VE table?!

https://rusefi.com/forum/viewtopic.php?f

rusefillc, 2021-09-21:
@OrchardPerformance @mck1117 what additional data would help here? I am planning to pay @Abricosvw a visit soon (he does not know yet but I know where he lives and I believe he does not lock his car)

OrchardPerformance, 2021-09-21:
Sensor type,part number and wiring/pinout. 
See if you can test the issue in some way with a bike pump or vacuum tester while the engine is off.