rusefi/rusefi #3264: all 4 of the mcu leds are on solid instead of rapidly flashing

rusefillc, 2021-09-25:
We have three reports of "4 of the mcu leds are on solid"

1) I had this on hellen-merc with https://github.com/rusefi/rusefi/releases/tag/2021.09.09_release
2) I had this on hellen-nb2

and now we have https://rusefi.com/forum/viewtopic.php?f=19&t=2131

rusefillc, 2021-09-25:
Maybe all we need is #3266

rusefillc, 2021-09-26:
Reopening due to complaint from Keith which at the moment REJECTS the "user has flashed wrong firmware" theory

```
As a short summary though, I was doing a few stops/starts with the ECU connected to my laptop. The ECU stopped communicating to the laptop, wouldn't start or anything. I pulled the cover noticed that the LEDs next to the STM were all solid which it doesn't do when its properly booted and functioning. After trying a bunch of times to get my computer to see it in DFU I reflashed the firmware and now it is connecting to tunerstudio again.
```

Side note to be tracked outside of this ticket - why the hell would it take multiple takes to enter DFU?

billstewartx, 2021-09-26:
the problem with the board bricking is a known issue---
try this-- have the board plugged into your laptop, via USB
which is powered on continuously
then turn on 12V after, (like your Key ON)
the board can freak out
try it for a few cycles, and see if it does it for you
That freakout happens to me
since i have a Raspberry Pi and screen plugged in continuously to Power.  and a USB connection to the board.
i always unplug the usb, turn on 12V, then plug in usb.

rusefillc, 2021-11-27:
looks like that's about wrong bundle, this was improved by bundle check on update in rusEFI console

closed 2021-11-27 by rusefillc