rusefi/rusefi #3265: HW CI on Hellen

rusefillc, 2021-09-25:
@BeerMoneyMotorsports #3264 is screaming "we heed HW CI on Hellen"

At the moment I have hellen-merc to put on this duty, open question what PC would be running CI.

Open question if @mck1117 would be interested

Maybe time to ask @ksmola if he plans to use his hellen121vag or if he should ship it to @mck1117 for HW CI duty.

ksmola, 2021-09-27:
I was planning on testing hellen121vag on E30, but If that's little value/low priority I can ship to Matt.

rusefillc, 2021-09-27:
@ksmola as long as there is a 40% probability of you using it in the next 6 months, please keep it.

mck1117, 2021-09-28:
The current hw ci has almost nothing to do with running on an actual hellen.  It would run on a disco board just fine.