rusefi/rusefi #3268: cranking is sensitive to WBO warm-up

rusefillc, 2021-09-26:
Visiting Nick today

first cranking attempt:
insert key, start cranking right away while external WBO controller gauge shows 14 which maybe probably means "heating"
result: car was very unhappy and did not start

second cranking attempt:
insert key, turn into ON position, give it 10 seconds for WBO to change from 14 to 21, car starts much better

question: is this about Nick's tune depending on closed loop fuel adjustment for cranking/running?
is this about Nick's battery being unhappy during sensor heating?
is this about anything else?

mck1117, 2021-09-28:
Is closed loop fuel enabled?

rusefillc, 2021-09-28:
@mck1117 i believe it's disabled for now https://rusefi.com/online/view.php?msq=654

mck1117, 2021-11-29:
we certainly don't listen to WBO while cranking

closed 2021-11-29 by mck1117