rusefi/rusefi #3273: why do we even allow users to enter VE of 170%?

rusefillc, 2021-09-26:
if any VE above 110% is totally unrealistic should we cap VE table entries at 150% just to allow some minor tune defects but not the totally crazy stuff?

OrchardPerformance, 2021-09-26:
Tempted to agree, there is no need to input above that, even 130% implies a problem with the setup but sometimes you do have to bodge a bit.

stefanst, 2021-09-26:
I guess that is important for auto-tune. If you manually enter a number >150 or so, let's assume you know what you're doing.
While autotune does not specifically allow to set min/max values for cells, it does already have tools to limit changes.

rusefillc, 2021-09-26:
@stefanst we limit unreasonable values in some places, where is the red line of crazy we allow VS crazy we shall not allow?

OrchardPerformance, 2021-09-27:
Can we allow over 110% but give a warning? 
Disallow anything over 130%?

If you are having to put in greater than 130 your bodging around another problem and should fix the problem before trying to tune. (Hell even greater than 100 but sometimes you can't avoid it).

OrchardPerformance, 2021-09-27:
This is also why autotune is the devil, it will happily wind up your VE values to stupid amounts ignoring that you have other problems. Thus sending you into a death spiral of bad tuning.

rusefillc, 2021-09-27:
Related #3277