rusefi/rusefi #3274: is it possible to validate injector flow on a running vehicle somehow?

rusefillc, 2021-09-26:
problem statement: we do not trust random injector random flow number, we do not trust dead time

for instance if we command firmware to add artificial 0.1ms of injection time

we should be able to calculate extra fuel amount
knowing engine displacement etc can we calculate theoretical AFR effect of extra fuel and compare it with actual measured AFR?

or maybe manually add 10% VE and compare measured AFR change with theoretical AFR change?

rusefillc, 2021-09-26:
@dron0gus @stefanst @OrchardPerformance @blundar what do you guys think?

OrchardPerformance, 2021-09-26:
This is similar to the dead time simultaneous/batch method to switch between the two and see if the AFR changes. 
Yes I think it is possible but the question becomes how to apply it to a real world engine running situation without other confounding factors making accuracy impossible

rusefillc, 2021-09-26:
@OrchardPerformance i think we need to compare actual change of AFR vs theoretical on a perfectly tuned engine with 100% known calibrations of everything to get a baseline of expected error, vs same error on a shitty calibration data?

rusefillc, 2021-09-26:
if anyone is interested to play with math attached is the log where we change flow rate from 770 to 700  to 630 to 570 cc/min
[2021-09-26_flow-rate-770-700-630-570.zip](https://github.com/rusefi/rusefi/files/7232463/2021-09-26_flow-rate-770-700-630-570.zip)

630
[image]

700
[image]

OrchardPerformance, 2021-09-27:
IMO right now the best way to determine this is to get a set of graduated containers from ebay and put injectors pointing into them and actually measure the flow rate. 
There are a couple of posts on the forum how to do this. 

Doing it on engine begs a few questions - when do we intend to do it? Can't do it while driving as transients will make an utter mess of the results as the afr changes due to enrichment and acceleration compensation. 
If we do it at idle the engine will change idle speed as it gians and looses power with the fuel mixture change, again messing up the accuracy of the results. 

Could we ditter the flow rate over a long period of time when the engine is low load steady state cruise? Maybe, that might just work for detecting small errors in the flow rate but it would be highly dependent on having the dead times set correctly otherwise this would totally mess up the results. 

If you have some unknown injectors the best thing to do is either get known injectors or flow test them manually.

rusefillc, 2021-09-27:
@OrchardPerformance i am looking for a killer feature and was wondering if this is a low hanging fruit of at least somewhat useable injector flow validator

a test procedure any Joe can do in 4 minutes would be the dream

OrchardPerformance, 2021-09-27:
It's a "killer feature" but it is probably far from low hanging.

blundar, 2021-09-27:
You need a leg to stand on for this to work, i.e. you need to know airflow
REALLY well and have spark well enough to not have observed afr deflection
as a result of spark.  If your airflow model is garbage and you double
fueling, you have double garbage air fuel ratio.

pk386, 2021-09-28:
Superfastmatt gives a method for testing fuel injectors. You have to pull the fuel rail though.
(Need to do this to my car....)

https://youtu.be/9ALDDK5u5Q8?t=145

mck1117, 2021-09-28:
Yeah, you can't really measure injector flow on-vehicle UNLESS the VE table is already very well validated.  If your injectors are 10% off, you can fix it in the VE table, so everything *looks* fine, but is actually 10% off.  Ask me how I know, my injectors were set 12% too low, but everything worked just fine because the VE table was also 12% low to compensate.  I ultimately corrected the wrong napkin math from 5 years ago by testing my injectors the old fashioned way with a plastic bottle and scale, then firing some known pulses in.