rusefi/rusefi #3276: Automatic dead time tuning?

OrchardPerformance, 2021-09-27:
As the inverse of the suggested flow rate test the dead time test works because we are trying not to have any change of afr when we switch from sequential to batch fuel and back. 
Batch multiplies the influence of dead time so incorrect dead time will produce a richer afr when too long and leaner when too short. 

We should be able to auto tune this at idle as we are trying to reach a point where the engine does not change state. 

At minimum could we do a button to make the ecu switch between the fueling modes 5 times when pressed at idle speed? 

Low priority feature.

rusefillc, 2021-09-27:
Dup of #492?

OrchardPerformance, 2021-09-27:
Looks like it, was sure we had one already but could not see it. Closed.

closed 2021-09-27 by OrchardPerformance