rusefi/rusefi #3278: How exactly did I brick Nick's Frankenso?

rusefillc, 2021-09-27:
I think I've bricked Nick's Frankenso yesterday

Frankenso in the car, long USB cable with extension, dry day, laptop running on battery

I was trying https://github.com/rusefi/rusefi/issues/3266 but in my recollection DFU process has not even started due to bug in console compatibility verification. I believe that i did NOT flash wrong firmware or initiate any flashing at all.

We have recovered ECU via Frankenso build-in st-link and did not have this problem repeating on the same day

(Nick's Frankenso only powers via +12 from car, it does not power on via USB - hmm why is that by the way?)

rusefillc, 2021-09-27:
i was 100% NOT in TS

mck1117, 2021-09-28:
can you get this problem to repeat, then dump the chip to a .bin file via stlink?

rusefillc, 2021-09-28:
Making a mental note to make a .bin dump next time I brick any ECU

I am available to see @Abricosvw today or tomorrow if he would be happy to see my face

rusefillc, 2021-09-28:
we have questionable login in FW:

read flash on start, reset if invalid. could weak power cause bad read and bad write?

should we change this logic?

mck1117, 2021-09-28:
Wait, does it really re-write a new config on first boot? I thought it just reset defaults (in RAM only!) if flash CRC didn't match.

rusefillc, 2021-09-28:
worth double checking

rusefillc, 2021-09-29:
@mck1117 you are right, it only resets into RAM and does not write to flash

closed 2025-08-10 by rusefillc