rusefi/rusefi #3284: individual cylinder fuel trims

rusefillc, 2021-09-30:
@blundar need more details on this feature

would it be in % extra, positive or negative, with "0" meaning no trim? or anything else?

rusefillc, 2021-09-30:
@mck1117 says something like "give #7 5% more gas"

blundar, 2021-09-30:
I don't think it really matters as long as it is clear.

I think I personally prefer 1.0 = no change, 0.9 = 10% reduction, 1.1 = 10%
increase
0 = no change +10 = 10% increase, -10 = 10% decrease would be ok

rusefillc, 2021-09-30:
fancy boys have 8x12 tables for each cylinder

rusefillc, 2021-09-30:
[image]

rusefillc, 2021-10-02:
Status: configuration maybe is ready, TS view maybe ready, no code.

mck1117, 2021-12-20:
@blundar says

> if you can set your axes points, 4x4 is plenty
> 4x4 lets you exclude idle, exclude cruise/off boost
> that's 110% functional.

closed 2022-01-01 by mck1117