rusefi/rusefi #3286: Drivers not work in W10

MIcQo, 2021-09-30:
Hi, im using microRusEfi.

I tried install all required drivers on windows 10 without success.

After installing it from bundle, windows still thinks to STM32 Bootloader (looks like not installed driver) instead of the STM Device in DFU Mode.

```
ERROR: Maybe DFU device not attached? Please check Device Manager.
ERROR: Maybe ST DFU Driver is missing?
ERROR: Maybe driver conflict with STM32Cube?
ERROR: Reminder about 'Install Drivers' button on top of rusEFI splash screen
Windows 10 10.0 amd64
```

No STM32Cube installed.

DFU drivers used from bundle.
Device is connected in DFU mode.

Please help, I'm definitely desperate.

rusefillc, 2021-09-30:
please share image of device manager

MIcQo, 2021-09-30:
[image]

here is it

rusefillc, 2021-10-01:
I know we had similar troubles reported - we were failing to remove that driver :( do not know how this was resolved. maybe some low level driver management tool for Windows?

MIcQo, 2021-10-03:
Hi, i found drivers that works as well.

[ST Microelectronics-3.0.4.0.zip](https://github.com/rusefi/rusefi/files/7273342/ST.Microelectronics-3.0.4.0.zip)
.

closed 2021-10-03 by MIcQo