rusefi/rusefi #3300: flash_reboot_dfu.bat feedback

rusefillc, 2021-10-03:
calls the console, which reboots the ECU, but then does not complete and freezes in anticipation of something, without closing the port - and no further dfu firmware update occurs

rusefillc, 2021-10-03:
[image]

rusefillc, 2021-10-03:
[image]

rusefillc, 2021-10-03:
@andreika-git unable to reproduce, looks like expected in master with discovery

[image]

andreika-git, 2021-10-04:
Maybe it's somehow related to multiple COM ports?
[image]

rusefillc, 2021-10-05:
I've added some logging can you please try again with latest console in 20+ minutes from now?

rusefillc, 2021-10-05:
also your console is 0523 who knows what else has changed worth trying latest just to be sure

DFU reboot protocol should be totally the same

andreika-git, 2021-10-05:
[image]

andreika-git, 2021-10-06:
[image]

andreika-git, 2021-10-06:
It kinda works... It resets DFU and after ~8 seconds of delay it exits.

closed 2021-10-06 by rusefillc