rusefi/rusefi #3301: H7 fast ADC

rusefillc, 2021-10-03:
We need fast ADC in H7

rusefillc, 2021-10-03:
@mck1117 is driving on H7 with only slow ADC to the best of my knowledge

@andreika-git is interested in fast ADC looks like has has custom changes we need to coordinate upstream merge tactics

andreika-git, 2021-10-03:
Related to: https://github.com/rusefi/rusefi/pull/3031

mck1117, 2021-10-03:
H7 is currently actually using fast ADC exclusively - it just needs a bit of plumbing to make it work.

rusefillc, 2021-10-03:
@mck1117 we have
```
#define EFI_USE_FAST_ADC FALSE
```
is it just a confusing way to say "we use only one kind of ADC speed and it's actually fast but flag name is not great"?

mck1117, 2021-10-06:
https://github.com/rusefi/rusefi/blob/b5b86ac584253aeda8c4f98db571a672d82eaefd/firmware/hw_layer/ports/stm32/stm32_adc_v4.cpp#L110

rusefillc, 2021-10-11:
@mck1117 do you plan more changes in this area?

mck1117, 2021-10-11:
I think it's probably OK for now.  Normal high speed ADC functions work fine!

rusefillc, 2021-10-11:
@andreika-git what's the next step for this ticket in your opinion?

closed 2025-07-19 by rusefillc