rusefi/rusefi #3302: USB Console thread runs out of stack and corrupts surrounding

dron0gus, 2021-10-03:
rusefi.map:
```
 .bss._ZL10usbChannel
                0x0000000020045e94      0x12c /tmp/rusefi.elf.QVDVPE.ltrans0.ltrans.o
 .bss._ZL10usbConsole
                0x0000000020045fc0      0x4d0 /tmp/rusefi.elf.QVDVPE.ltrans0.ltrans.o
```
gdb:
```
(gdb) p/x usbChannel
$23 = {<TsChannelBase> = {_vptr.TsChannelBase = 0x80ad878 <vtable for UsbChannel+8>, scratchBuffer = {0x0 <repeats 286 times>}, name = 0x80ad850}, m_channel = 0x200228e8}
(gdb) p &usbChannel.m_channel
$1 = (BaseChannel * const *) 0x20045fbc <usbChannel+296>
(gdb) watch *0x20045fbc
Hardware watchpoint 1: *0x20045fbc
... let it run and connect to TS

Thread 5 received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 537158712]
Sensor::get (type=SensorType::Rpm) at ./controllers/sensors/sensor.cpp:202
202		const auto entry = getEntryForType(type);
(gdb) bt
#0  Sensor::get (type=SensorType::Rpm) at ./controllers/sensors/sensor.cpp:202
#1  0x00264c20 in updateTunerStudioState (tsOutputChannels=0x20045cd8 <tsOutputChannels>) at ./console/status_loop.cpp:464
#2  0x00266d28 in prepareTunerStudioOutputs () at ./console/status_loop.cpp:878
#3  0x00264274 in TunerStudio::cmdOutputChannels (this=0x200219a0 <tsInstance>, tsChannel=0x20045e94 <usbChannel>, offset=0, count=256) at ./console/binary/tunerstudio_commands.cpp:24
#4  0x00263ea4 in TunerStudioBase::handleCrcCommand (this=0x200219a0 <tsInstance>, tsChannel=0x20045e94 <usbChannel>, data=0x20045e99 <usbChannel+5> "", incomingPacketSize=5)
    at ./console/binary/tunerstudio.cpp:704
#5  0x00263330 in tsProcessOne (tsChannel=0x20045e94 <usbChannel>) at ./console/binary/tunerstudio.cpp:528
#6  0x002633d2 in TunerstudioThread::ThreadTask (this=0x20045fc0 <usbConsole>) at ./console/binary/tunerstudio.cpp:548
#7  0x00267d74 in ThreadController<800>::StaticThreadTaskAdapter (thread=0x20045fc0 <usbConsole>) at ./controllers/system/thread_controller.h:36
#8  0x0020035e in _port_thread_start () at ChibiOS/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S:201
(gdb) info reg
r0             0x20046018          537157656
r1             0x3                 3
r2             0x20045ca8          537156776
r3             0x3                 3
r4             0x264221            2507297
r5             0x20045fc0          537157568
r6             0x0                 0
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0x34d569            3462505
sp             0x20045fb8          **0x20045fb8 <usbChannel+292>**
lr             0x264c21            2509857
pc             0x26927c            0x26927c <Sensor::get(SensorType)+12>
xPSR           0x1000000           16777216
fpscr          0x0                 0
msp            0x20021000          0x20021000
psp            0x20045fb8          0x20045fb8 <usbChannel+292>
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x6                 6
(gdb) p/x usbChannel
$5 = {<TsChannelBase> = {_vptr.TsChannelBase = 0x80ad878 <vtable for UsbChannel+8>, scratchBuffer = {0x4f, 0x0, 0x0, 0x0, 0x1, 0x6b, 0x86, 0x8, 0x13, 0x8a, 0x4b, 0x32, 0x30, 0x32, 0x31, 0x31, 0x30, 0x30, 0x33, 
      0x40, 0x32, 0x38, 0x39, 0x31, 0x39, 0x0, 0x2c, 0xd6, 0x8, 0xcb, 0x33, 0x33, 0x5f, 0x66, 0x37, 0x2e, 0x33, 0x31, 0x34, 0x37, 0x36, 0x32, 0x34, 0x35, 0x36, 0x32, 0x0, 0x47, 0xef, 0xe7, 0xc5, 
      0x0 <repeats 235 times>}, name = 0x30ad850}, m_channel = 0x20046018}
```
usbChannel corrupted. And hardFault catched later:
```
Thread 5 received signal SIGTRAP, Trace/breakpoint trap.
HardFault_Handler () at main_hardfault.c:57
57	    bkpt();
(gdb) bt
#0  HardFault_Handler () at main_hardfault.c:57
#1  <signal handler called>
#2  0x00267c00 in UsbChannel::write (this=0x20045e94 <usbChannel>, buffer=0x20045e98 <usbChannel+4> "\001\001", size=263) at ./console/usb_console.cpp:27
#3  0x002626fe in TsChannelBase::writeCrcPacketSmall (this=0x20045e94 <usbChannel>, responseCode=0 '\000', buf=0x20045cd8 <tsOutputChannels> "\006", size=256) at ./console/binary/tunerstudio_io.cpp:48
#4  0x002627f0 in TsChannelBase::writeCrcPacket (this=0x20045e94 <usbChannel>, responseCode=0 '\000', buf=0x20045cd8 <tsOutputChannels> "\006", size=256) at ./console/binary/tunerstudio_io.cpp:91
#5  0x00262844 in TsChannelBase::sendResponse (this=0x20045e94 <usbChannel>, mode=TS_CRC, buffer=0x20045cd8 <tsOutputChannels> "\006", size=256) at ./console/binary/tunerstudio_io.cpp:101
#6  0x00264288 in TunerStudio::cmdOutputChannels (this=0x200219a0 <tsInstance>, tsChannel=0x20045e94 <usbChannel>, offset=0, count=256) at ./console/binary/tunerstudio_commands.cpp:26
#7  0x00263ea4 in TunerStudioBase::handleCrcCommand (this=0x200219a0 <tsInstance>, tsChannel=0x20045e94 <usbChannel>, data=0x20045e99 <usbChannel+5> "\001", incomingPacketSize=5)
    at ./console/binary/tunerstudio.cpp:704
#8  0x00263330 in tsProcessOne (tsChannel=0x20045e94 <usbChannel>) at ./console/binary/tunerstudio.cpp:528
#9  0x002633d2 in TunerstudioThread::ThreadTask (this=0x20045fc0 <usbConsole>) at ./console/binary/tunerstudio.cpp:548
#10 0x00267d74 in ThreadController<800>::StaticThreadTaskAdapter (thread=0x20045fc0 <usbConsole>) at ./controllers/system/thread_controller.h:36
#11 0x0020035e in _port_thread_start () at ChibiOS/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S:201
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) i local
ctx = {r0 = 537157268, r1 = 537157272, r2 = 263, r3 = 1088841318, r12 = 3462505, lr_thd = 2500351, pc = 2522112, xpsr = 16777216, s0 = 1052355483, s1 = 2150977, s2 = 0, s3 = 0, s4 = 0, s5 = 2387015, s6 = 0, 
  s7 = 788529152, s8 = 1122369536, s9 = 2143289344, s10 = 1120403456, s11 = 0, s12 = 2147483665, s13 = 0, s14 = 0, s15 = 4294967295, fpscr = 2217901101, reserved = 537157274}
faultType = HardFault
faultAddress = 0
isFaultPrecise = false
isFaultImprecise = false
isFaultOnUnstacking = false
isFaultOnStacking = false
isFaultAddressValid = false
(gdb) p/x ctx
$6 = {r0 = 0x20045e94, r1 = 0x20045e98, r2 = 0x107, r3 = 0x40e66666, r12 = 0x34d569, lr_thd = 0x2626ff, pc = 0x267c00, xpsr = 0x1000000, s0 = 0x3eb9ab9b, s1 = 0x20d241, s2 = 0x0, s3 = 0x0, s4 = 0x0, 
  s5 = 0x246c47, s6 = 0x0, s7 = 0x2f000000, s8 = 0x42e60000, s9 = 0x7fc00000, s10 = 0x42c80000, s11 = 0x0, s12 = 0x80000011, s13 = 0x0, s14 = 0x0, s15 = 0xffffffff, fpscr = 0x84327c2d, reserved = 0x20045e9a}
```
Currently "USB Console" has `CONNECTIVITY_THREAD_STACK` size stack. Equal to `(2 * UTILITY_THREAD_STACK_SIZE)` = 2 * 400 bytes.

dron0gus, 2021-10-03:
Week ago I got similar out of stack issue with "flash writer" thread and MFS (Managed Flash Storage) driver. I thought it is related to MFS use too much stack. But it was definitely ok with 400 bytes of stack when I initially PR'ed MFS support.
Were there any changes that can affect all tasks stack usage?

dron0gus, 2021-10-03:
Have no idea why `updateTunerStudioState` needs 0x370 (880) bytes of stack.
```
(gdb) frame 0
#0  Sensor::get (type=SensorType::Rpm) at ./controllers/sensors/sensor.cpp:202
202		const auto entry = getEntryForType(type);
(gdb) info reg sp
sp             0x20045fb8          0x20045fb8 <usbChannel+292>
(gdb) frame 1
#1  0x00264c20 in updateTunerStudioState (tsOutputChannels=0x20045cd8 <tsOutputChannels>) at ./console/status_loop.cpp:464
464		int rpm = Sensor::get(SensorType::Rpm).Value;
(gdb) info reg sp
sp             0x20045fd0          0x20045fd0 <usbConsole+16>
(gdb) frame 2
#2  0x00266d28 in prepareTunerStudioOutputs () at ./console/status_loop.cpp:878
878		updateTunerStudioState(&tsOutputChannels PASS_ENGINE_PARAMETER_SUFFIX);
(gdb) info reg sp
sp             0x20046340          0x20046340 <usbConsole+896>
```

dron0gus, 2021-10-03:
`updateTunerStudioState` prologue
```
0x00264c06	462	void updateTunerStudioState(TunerStudioOutputChannels *tsOutputChannels DECLARE_ENGINE_PARAMETER_SUFFIX) {
   0x00264c00 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+0>:	00 b5	push	{lr}
   0x00264c02 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+2>:	2d ed 02 8b	vpush	{d8}
=> 0x00264c06 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+6>:	ad f5 59 7d	sub.w	sp, sp, #868	; 0x364
   0x00264c0a <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+10>:	d8 ab	add	r3, sp, #864	; 0x360
   0x00264c0c <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+12>:	a3 f5 55 73	sub.w	r3, r3, #852	; 0x354
   0x00264c10 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+16>:	18 60	str	r0, [r3, #0]

```

mck1117, 2021-10-03:
1. I have no idea why it needs 880 bytes of stack either.
2. `CONNECTIVITY_THREAD_STACK` was recently decreased from 3x to 2x `UTILITY_THREAD_STACK_SIZE`, so that should maybe go back up (at least for now)

mck1117, 2021-10-03:
Ok, I think the problem is that it's just a huge function, and when optimization is disabled, it uses a hilarious amount of stack space for all the locals and scratch calculations.  When optimization is enabled, it reuses the same space on the stack over and over again, but without optimization it piles up quickly.

rusefillc, 2021-10-03:
@dron0gus we have progress under #3308 can you please try again?

dron0gus, 2021-10-03:
1st problem - I'm getting hard fault with high optimization too.
2nd problem is that I'm not able to debug with high optimization. Also linker re-arranges objects and something else gets corrupted.

I'm not sure, why 792 bytes are allocated on stack and passed through r0 to `Sensor::get(SensorType)`. I understand how asm is generated from plain C code. But have no idea about C++. And I'm not sure that optimization can reduce arguments size.
```
=> 0x00264c12 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+18>:	d8 ab	add	r3, sp, #864	; 0x360
   0x00264c14 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+20>:	a3 f5 46 73	sub.w	r3, r3, #792	; 0x318
   0x00264c18 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+24>:	03 21	movs	r1, #3
   0x00264c1a <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+26>:	18 46	mov	r0, r3
   0x00264c1c <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+28>:	04 f0 28 fb	bl	0x269270 <_ZN6Sensor3getE10SensorType>

```

dron0gus, 2021-10-03:
@rusefillc I'll try.

dron0gus, 2021-10-03:
**-O0** optimization - same Hard Fault.

dron0gus, 2021-10-03:
```
0x002659c2	662		int rpm = Sensor::get(SensorType::Rpm).Value;
   0x002659c0 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+16>:	66 ab	add	r3, sp, #408	; 0x198
=> 0x002659c2 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+18>:	a3 f5 c2 73	sub.w	r3, r3, #388	; 0x184
   0x002659c6 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+22>:	03 21	movs	r1, #3
   0x002659c8 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+24>:	18 46	mov	r0, r3
   0x002659ca <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+26>:	03 f0 19 fb	bl	0x269000 <_ZN6Sensor3getE10SensorType>
```

mck1117, 2021-10-04:
>  I understand how asm is generated from plain C code. But have no idea about C++. And I'm not sure that optimization can reduce arguments size.

C++ assembly is generally the same as C code, but when instance member functions (ie, not static) are called, an additional "zeroth" argument is added to pass the `this` pointer.  Other than that, the calling convention and everything are the same as C code.

> why 792 bytes are allocated on stack and passed through r0 to Sensor::get(SensorType)

That function returns a struct, and when the struct doesn't fit in a register, you pass instead a pointer to a place on the stack for the callee to place the result.  It's using `r0` as the return pointer (hence the pointer in r0 to somewhere on the stack), and `r1` as the argument `type`.  I believe the same thing happens in C when you try to return struct where `sizeof(S) > sizeof(uintptr_t)`

[image]

mck1117, 2021-10-04:
Ok, yep, went and looked it up, and that is indeed the case.

> A Composite Type larger than 4 bytes, or whose size cannot be determined statically by both caller and callee, is stored in memory at an address passed as an extra argument when the function was called (Parameter Passing, rule A.4). The memory to be used for the result may be modified at any point during the function call.

https://developer.arm.com/documentation/ihi0042/j/?lang=en#result-return

mck1117, 2021-10-04:
So passing that extra parameter to the `Sensor::get` function is entirely normal, it looks like.

mck1117, 2021-10-04:
Ohh, part of the reason I didn't see this as much is that it looks like GCC is more efficient while in cm7 mode (I was debugging on H7) than it is on cm4 (based on memory addresses, it looks like you're debugging against an STM32F4).  It uses more stack on my F4 build.

dron0gus, 2021-10-05:
https://en.cppreference.com/w/cpp/language/lifetime **Temporary object lifetime**
So each `Sensor::get(SensorType::xxx).value_or(0)` call creates temporary `SensorResult` who's lifetime lasts until end of function.

dron0gus, 2021-10-05:
```
662		int rpm = Sensor::get(SensorType::Rpm).Value;
=> 0x002659f0 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+16>:	66 ab	add	r3, sp, #408	; 0x198
   0x002659f2 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+18>:	a3 f5 c2 73	sub.w	r3, r3, #388	; 0x184
   0x002659f6 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+22>:	03 21	movs	r1, #3
   0x002659f8 <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+24>:	18 46	mov	r0, r3
   0x002659fa <_Z22updateTunerStudioStateP25TunerStudioOutputChannels+26>:	03 f0 19 fb	bl	0x269030 <_ZN6Sensor3getE10SensorType>
```
Still 408 bytes on stack.

mck1117, 2021-10-05:
>  Looks compiler optimization can do nothing with this.

This is not quite true - the compiler is free to do anything that has no observable result.  Since `SensorResult` doesn't have a destructor, the destruction of a `SensorResult` has no observable effect, so the compiler is allowed to *skip it completely*.  When optimization is enabled, it will indeed destruct those instances early and recycle the stack space for the next one.  At `-O2`, it's using a tiny fraction of the stack space that it does at `-O0`.

mck1117, 2021-10-05:
With optimization turned on, but LTO turned off (so it doesn't inline everything), `updateTunerStudioState` only allocates itself 52 bytes of stack.

mck1117, 2021-10-05:
> call creates temporary `SensorResult` who's lifetime lasts until end of function.

Oh, also, this isn't true either.  Temporaries are destructed at the end of the *line*, not at the end of the scope.  The temporary is destructed as part of the semicolon at the end of the line, prior to the next line executing.

rusefillc, 2021-10-05:
@dron0gus please try again now with latest master?

mck1117, 2021-10-05:
`-fstack-usage` is a potentially useful option, spits out files that look like this:

```
status_loop.cpp:462:13:void updateTempSensors()	56	static
status_loop.cpp:478:13:void updateThrottles()	48	static
status_loop.cpp:495:13:void updateLambda()	48	static
status_loop.cpp:505:13:void updateFuelSensors()	56	static
status_loop.cpp:516:13:void updateVvtSensors()	24	static
status_loop.cpp:526:13:void updateVehicleSpeed(int)	40	static
status_loop.cpp:534:13:void updateRawSensors()	64	static
status_loop.cpp:551:13:void updatePressures()	48	static
status_loop.cpp:557:13:void updateMiscSensors()	56	static
status_loop.cpp:572:13:void updateSensors(int)	16	static
status_loop.cpp:584:13:void updateFuelCorrections()	32	static
status_loop.cpp:595:13:void updateFuelLoads()	24	static
status_loop.cpp:602:13:void updateFuelResults()	32	static
status_loop.cpp:613:13:void updateFuelInfo()	40	static
status_loop.cpp:631:13:void updateIgnition(int)	32	static
status_loop.cpp:640:13:void updateFlags()	8	static
status_loop.cpp:666:13:void updateTpsDebug()	120	static
status_loop.cpp:680:6:void updateTunerStudioState(TunerStudioOutputChannels*)	240	static
```

rusefillc, 2021-10-19:
@dron0gus poke

rusefillc, 2021-11-09:
@dron0gus is this still an issue or not?

@racer-coder just FYI

dron0gus, 2021-11-09:
Not an issue after @mck1117 fixes.

closed 2021-11-09 by mck1117