rusefi/rusefi #3305: rusefi_Console: Too long re-connection time

andreika-git, 2021-10-03:
TunerStudio is able to restore the connection in a few seconds after the USB cable is connected again.
But the Console has a too long reconnection timeout (>10 seconds?). It would be perfect if we could match the TS timings.

mck1117, 2021-10-03:
>  But the Console has a too long reconnection timeout (>10 seconds?). It would be perfect if we could match the TS timings.

That is, if it works at all. Usually the console just crashes for me upon reconnect attempt.

rusefillc, 2021-10-06:
Q: what is this about?
A: @andreika-git often need to reset ECU while bench testing stuff like stepper etc so we want a lighter ECU reset workflow

rusefillc, 2021-12-06:
@alxrMironovHome can you help?

rusefillc, 2022-01-12:
@andreika-git how does it look with the change?

rusefillc, 2022-01-12:
@alxrMironovHome discovered that drama was about poking driver with port number which was not yet valid at that point. that was hanging-ish the driver.

now with the fix we wait for port to become officially available again and things seem to have improved!

mck1117, 2022-02-16:
Fixed by #3936

closed 2022-02-16 by mck1117