rusefi/rusefi #3309: why does averagedMapRunningBuffer use min() instead of avg()

mck1117, 2021-10-03:
Why does `averagedMapRunningBuffer` use `min(...)` instead of `avg(...)`?

I think I understand what the feature is for, to average the sampled MAP across multiple sample events (ie, cylinders) to get a smoother reading.  But why would it select the minimum of all samples, instead of the average of all samples?

andreika-git, 2021-10-04:
That's because it is used mostly for ITBs. Each ITB has its own vacuum tube, but they are interconnected into one common MAP input. And only one cylinder has valid MAP readings at a time - one ITB draws-in air (intake stroke), but others can push out air because of resonant effects in the manifold (the inlet valve closes and the air wave moves backward). In other words, the MAP sensor value can be greater than the real cylinder vacuum - and it means that the bigger value is wrong. And at the same time MAP readings cannot be less than the real vacuum - it means we can take the smallest one as the "true" one!
Here's the example of a smooth ITB MAP data:
https://rusefi.com/forum/download/file.php?id=3154
It was a major breakthrough for me back in 2017 :)

mck1117, 2021-10-04:
Do you have any data to support that different values of `mapMinBufferLength` actually provide a difference in the measured value or smoothness?

mck1117, 2021-10-04:
Also, remember that especially for ITBs, the goal is not *correct*, but instead *repeatable*, and average may be a more *repeatable* measure that will also give better transient response (the whole point of ITBs!).

andreika-git, 2021-10-04:
> Do you have any data to support that different values of `mapMinBufferLength` actually provide a difference in the measured value or smoothness?

I use only mapMinBufferLength = numberOfCylinders (meaning the last N cylinders, N=4 in my case).
I've also tested values less than numberOfCylinders and it didn't work well for me. And of course = 0 (or =1) just disables the feature. 
All other larger values like 2*N etc. make no sense to me (= slower response).

mck1117, 2021-10-04:
Yes, I certainly agree that the only valid values are probably 1 or N=cylNum.  Do you have back to back comparisons of N=1 and N=4 on your engine?

andreika-git, 2021-10-04:
It was just unusable with N=1:
[image]
[image]

mck1117, 2021-10-04:
Wait, but that looks like a MAP sensor on only a single cylinder.

[image]

Once the engine is running the aliasing is pretty bad (TS update rate vs. MAP update rate), so it's hard to tell if it's really that noisy, or just measuring on a single cylinder.  If it's really that much variation in MAP between cylinders, I have to wonder about cylinder-to-cylinder balance (mechanical) and plumbing balance for the MAP sensor.

Anyway, if it was a small to moderate variation, average would work fine too.  Sort of hard to tell from that log how bad it really is.

andreika-git, 2021-10-04:
It's definitely not only 1 cylinder :)
There are 2 pairs of 4 hoses connected to common junctions:
[image]
[image]
One quadruple is for MAP sensor, another is for vacuum brake booster.
And the engine is idling rather nasty with N=1, trust me.

mck1117, 2021-10-04:
My question is not whether `N=1` is correct for your engine - my question is whether *average* instead of *minimum* would provide equivalent results.  A log at sufficient update rate with N=1 would provide enough information to make that determination.

mck1117, 2021-10-04:
PS: are those motorcycle throttles?

andreika-git, 2021-10-04:
> My question is not whether `N=1` is correct for your engine - my question is whether _average_ instead of _minimum_ would provide equivalent or superior results. A log at sufficient update rate with N=1 would provide enough information to make that determination.

I cannot find logs with the "avg" version, sorry. It was 4 years ago, and the "min" version appeared after I tried the "avg" first. Also I remember that current "min of the window-averaged samples" version was better than "min of all samples during the cycle".

> PS: are those motorcycle throttles?

yes, and they were balanced:
[image]

andreika-git, 2021-10-04:
Min. MAP value is a common practice for ITBs. The ideal solution is to use 4 separate MAP sensors and choose the min value. For example, @jharvey has a hardware solution for combining MAPs: https://github.com/jharvey/MultiMAP

Also people use back-flow valves as a mechanical substitute...

mck1117, 2021-10-04:
well, anyway, current behavior is preserved in #3292

closed 2021-11-29 by mck1117