rusefi/rusefi #3311: Fuel Consumption - Integration for BMW PNP

OrchardPerformance, 2021-10-03:
Now we have a working measure of fuel consumption there is a software task to provide a PWM output that matches the fuel rate. #45

It is believed that the stock ECU puts out a pulse that matches the fuel injector pulse. 

To achieve this on RE we need a pulse that is increased in duration proportionally to the injector pulse and had a modifier to calibrate the fuel rate when bigger injectors are used. 

TLDR: 
Create Pulse that matches Injector pulse
Have % pulse length modifier to adjust output pulse

mck1117, 2021-10-04:
luaaaaaaaaaa

OrchardPerformance, 2021-10-05:
Perhaps this is one to put into an Lua script but currently how does one Lua?

We have https://github.com/rusefi/rusefi/wiki/Lua-Scripting and https://rusefi.com/forum/viewtopic.php?f=5&t=2062

We have an "Lua Outputs" menu in TS but beyond that where does a _user_ Lua?