rusefi/rusefi #3341: fuel pressure and injector flow

rusefillc, 2021-10-09:
Let's explain this a bit better

[image]

huh? where did 379 come from?

Matthew Kennedy  4:52 PM
379 is 55 psi in kpa

andrey  4:53 PM
shall we rename to "fuel rail pressure">
4:53
?

4:53
for the sake of walking meat?
New

Matthew Kennedy  4:54 PM
we use that parameter for two things
4:54
it's always the pressure at which the injectors flow the specified amount
4:54
in case of sensed rail pressure, it's the "P1" in the math the ECU does
4:55
in case of fixed rail pressure, P2 is that value plus map

andrey  4:56 PM
``float fuelReferencePressure;+This is the pressure at which your injector flow is known.\nFor example if your injectors flow 400cc/min at 3.5 bar, enter 350kpa here.;
4:56
float fuelReferencePressure;+This is the pressure at which your injector flow is known.\nFor example if your injectors flow 400cc/min at 3.5 bar, enter 350kpa here.;
4:56
this gives a wrong impression in my opinion
4:56
 your injector flow is known
4:57
that part takes focus away from the fact that we need to know YOUR fuel pressure rail

Matthew Kennedy  4:57 PM
that tooltip is missing the info for "fixed rail pressure" mode
4:57
yeah

andrey  4:57 PM
also shall we use conditional TS magic here for label name?

OrchardPerformance, 2021-10-11:
"Injector pressure when tested"
"Injector test pressure"
"Flow test pressure"
"Injector calibration pressure" 
"Flow calibration pressure"

I feel like one of the last two?

duhafnusa4, 2021-11-04:
> "Injector pressure when tested" "Injector test pressure" "Flow test pressure" "Injector calibration pressure" "Flow calibration pressure"
> 
> I feel like one of the last two?

Injector Calibration Pressure with tooltip.

OrchardPerformance, 2021-11-05:
I can make that happen I think. 

Assuming we all agree?

rusefillc, 2021-11-05:
I agree with anything.

duhafnusa4, 2021-11-05:
@OrchardPerformance I think you have a blessing

rusefillc, 2022-12-11:
@mi-hol here's how those go :(

rusefillc, 2022-12-11:
@mi-hol I am trying to say that we get stuck in paralysis analysis or lack of urgency or life or else, and in the end I am not happy with the amount of documentation progress.

mi-hol, 2022-12-12:
another topic for a call